PaulStoffregen/FreqCount #4: Frequency spike for >5 Mhz measurements

kmcpeak, 2016-04-08:
Dear Paul,

I am using the FreqCount library with openQCM to measure the frequency of a 6Mhz crystal oscillator. When I run the openQCM code (see below) it works really well except for these strange glitches every 200 secs or so where the frequency spikes (see graph). The openQCM developer is saying this is a FreqCount issue that occurs for crystals > 5Mhz. Have you observed this behavior before for high freq measurements? The output in the image is simply the frequency every 1 sec, no averaging.

`/* LICENSE
 * Copyright (C) 2014 openQCM
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see http://www.gnu.org/licenses/gpl-3.0.txt
 *
 * INTRO
 * openQCM is the unique opensource quartz crystal microbalance http://openqcm.com/
 * openQCM Java software project is available on github repository 
 * https://github.com/marcomauro/openQCM
 * 
 * Measure QCM frequency using FreqCount library developed by Paul Stoffregen 
 * https://github.com/PaulStoffregen/FreqCount
 *
 * NOTE       - designed for 6 and 10 Mhz At-cut quartz crystal
 *            - 3.3 VDC supply voltage quartz crystal oscillator
 *            - Configure EXTERNAL reference voltage used for analog input
 *            - Thermistor temperature sensor
 *
 * author     Marco Mauro 
 * version    1.1
 * date       april 2015 
 *
 */

// include library for frequency counting
#include <FreqCount.h>

// fixed "gate interval" time for counting cycles 1000ms  
#define GATE   1000
// Thermistor pin
#define THERMISTORPIN A1 
// // resistance at 25 degrees C
#define THERMISTORNOMINAL 10000
// temp. for nominal resistance (almost always 25 C)
#define TEMPERATURENOMINAL 25   
// how many samples to take and average 
#define NUMSAMPLES 10
// The beta coefficient of the thermistor (usually 3000-4000)
#define BCOEFFICIENT 3950
// the value of the 'other' resistor
#define SERIESRESISTOR 10000    

// print data to serial port 
void dataPrint(unsigned long Count, int Temperature){
  //Serial.print("RAWMONITOR");
  Serial.println(Count);
  //Serial.print("_");
  //Serial.print(Temperature);
  //Serial.write(255);
}

// measure temperature
int getTemperature(void){
  int i;
  float average;
  int samples[NUMSAMPLES];
  float thermistorResistance;
  int Temperature; 

  // acquire N samples
  for (i=0; i< NUMSAMPLES; i++) {
    samples[i] = analogRead(THERMISTORPIN);
    delay(10);
  }

  // average all the samples out
  average = 0;
  for (i=0; i< NUMSAMPLES; i++) {
    average += samples[i];
  }
  average /= NUMSAMPLES;

  // convert the value to resistance
  thermistorResistance = average * SERIESRESISTOR / (1023 - average);
  
  float steinhart;
  steinhart = thermistorResistance / THERMISTORNOMINAL;          // (R/Ro)
  steinhart = log(steinhart);                       // ln(R/Ro)
  steinhart /= BCOEFFICIENT;                        // 1/B * ln(R/Ro)
  steinhart += 1.0 / (TEMPERATURENOMINAL + 273.15); // + (1/To)
  steinhart = 1.0 / steinhart;                      // Invert
  steinhart -= 273.15;                              // convert to C

  // decimal value
  Temperature = steinhart * 10;
  return(Temperature);
}

// QCM frequency by counting the number of pulses in a fixed time 
unsigned long frequency = 0;
// thermistor temperature
int temperature = 0;

void setup(){
  Serial.begin(115200);
  // Configure the reference voltage used for analog input 
  analogReference(EXTERNAL);
  // init the frequency counter
  FreqCount.begin(GATE);
}

void loop(){
  if (FreqCount.available()) 
  {
    frequency = FreqCount.read();       // measure QCM frequency
    temperature = getTemperature();     // measure temperature 
    dataPrint(frequency, temperature);  // print data
  }
}
`

Thanks a lot for your help!

Regards,
Kevin

[image]

PaulStoffregen, 2016-04-09:
Two questions:

Have you tried running FreqCount's Serial_Output example can capture it with a program like CoolTerm?

Which hardware are you using?  If I try to reproduce this (which would be much more likely if it's reproducible without the other software), this little detail matters greatly...

kmcpeak, 2016-04-10:
Paul,

Below is output from FreqCount's Serial_Output captured in TeraTerm and plotted up. This is running on an Arduino Micro (https://www.arduino.cc/en/Main/ArduinoBoardMicro), microcontroller: ATmega32U4. Thanks.

Regards,
Kevin

[image]

kmcpeak, 2016-04-10:
Paul,

To simplify the setup I used a function generator (Tektronix AFG3022) to input 1,5 and 6 Mhz into the Arduino Micro and an Agilent 53132A counter. Below are plots of FreqCount's Serial_Output and the Agilent counter. Notice the glitches at all frequencies for the Serial_Output. Can you please look into resolving this. Thank you.

Regards,
Kevin

[image]
[image]
[image]

PaulStoffregen, 2016-04-10:
Each glitch appears to be one higher measurement, followed by one balancing lower one.  That's what happens if the counter is read late.  The next count makes up the difference.

My guess is the timer0 interrupt for millis() is delaying the acquisition.  Please try adding this in setup()

      TIMSK0 &= ~(1 << TOIE0);   //  Disable Timer 0 interrupt

kmcpeak, 2016-04-11:
Paul,

Here is the code:
`/* FreqCount - Example with serial output
 * http://www.pjrc.com/teensy/td_libs_FreqCount.html
 *
 * This example code is in the public domain.
 */
#include <FreqCount.h>

void setup() {
  TIMSK0 &= ~(1 << TOIE0);   //  Disable Timer 0 interrupt
  Serial.begin(57600);
  FreqCount.begin(1000);
}

void loop() {
  if (FreqCount.available()) {
    unsigned long count = FreqCount.read();
    Serial.println(count);
  }
}
`
But the output still has glitches, see image.
[image]