PaulStoffregen/FreqCount #20: Arduino nano every support

tim-hamberg, 2021-02-10:
Hi,
Very usefull library! 
Is it possible to update the library so it works with the newer Arduino nano every?

PaulStoffregen, 2021-02-10:
Depends on the meaning of "possible".

If "possible" means Nano Every has good enough hardware and someone with enough knowledge and time creates and tests the code, then yes.  I'd be happy to merge a pull request, as long as it doesn't break compatibility for other boards, when/if someone contributes such code.

But if "possible" means I will do it all by myself, then no.

EdgeOg0322, 2021-08-09:
Is there a manual or explanation on how you've coded the program to work for other boards? I would like to contribute in making it compatible with nano every too.

PaulStoffregen, 2021-08-09:
No, there is no manual.  The only way to learn this is to read the source code.