PaulStoffregen/FreqCount #22: Error increases as frequency increases using FreqCount Library in Arduino Uno Board

SumanTheBob, 2021-08-13:
/* FreqCount - Example with LCD output
 * http://www.pjrc.com/teensy/td_libs_FreqCount.html
 *
 * This example code is in the public domain.
 */
#include <FreqCount.h>
#include <LiquidCrystal.h>

LiquidCrystal lcd(5, 4, 3, 2, 1, 0);

void setup() {
  Serial.begin(9600);
  lcd.begin(8, 2);
  lcd.print("Freq:");
  FreqCount.begin(1000);
}

void loop() {
  if (FreqCount.available()) {
    unsigned long count = FreqCount.read();
    lcd.setCursor(0, 1);
    lcd.print(count);
    lcd.print("       ");
  }
}
I am using this program for measuring the signal frequency of a square wave (8 v p-p) varying from 2 kHz to 10 kHz. But the error increases as the frequency increases. (For 2 kHz the LCD output is 2003 Hz, for 5 kHz it is 5006 and for 10 kHz the LCD output is 10011: hence the error is always positive and it increases with frequency)Can you please guide me on why this is happening?

colinluthier, 2023-02-04:
Are you feeding the clock of the Arduino with a calibrated source? Replacing the crystal with a TCXO will help with accuracy and stability. 
Alternatively you can add a bit of code for an offset for your particular crystal.

PaulStoffregen, 2023-02-04:
You say "the error increases as the frequency increases", but to my eyes the error looks like a pretty consistent percentage of the frequency measured.